rm-hull/luma.lcd #8: Some demos don't work with Pillow V3

DougieLawson, 2015-12-28:
```
root@saturn /tmp/pcd8544/examples # ./demo.py
Traceback (most recent call last):
  File "./demo.py", line 24, in <module>
    image.demo()
  File "/tmp/pcd8544/examples/image.py", line 8, in demo
    drawing()
  File "/tmp/pcd8544/examples/image.py", line 48, in drawing
    lcd.image(im)
  File "/usr/local/lib/python2.7/dist-packages/pcd8544/lcd.py", line 118, in image
    data([BITREVERSE[ord(x)] ^ mask for x in list(rim.tostring())])
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 686, in tostring
    "Please call tobytes() instead.")
Exception: tostring() has been removed. Please call tobytes() instead.

```

rm-hull, 2015-12-28:
Is Pillow v3 tied to python 3?

DougieLawson, 2015-12-29:
Nope. It's just what I got with a stock pip install.

rm-hull, 2015-12-30:
I was previously using pillow v1.7.6
Seems like `tostring()` is not exactly the same as `tobytes()` for 1-bit images

Still investigating this one...

rm-hull, 2017-01-14:
Project has had a major overhaul, and works directly with the latest pillow version (v4.0.0)

closed 2017-01-14 by rm-hull