rm-hull/luma.lcd #9: sys_info demo doesn't work with psutil 3.3.0

DougieLawson, 2015-12-28:
```
root@saturn /tmp/pcd8544/examples # ./sys_info.py
Traceback (most recent call last):
  File "./sys_info.py", line 78, in <module>
    main()
  File "./sys_info.py", line 75, in main
    stats()
  File "./sys_info.py", line 63, in stats
    lcd.text(cpu_usage())
  File "./sys_info.py", line 39, in cpu_usage
    uptime = datetime.now() - datetime.fromtimestamp(psutil.BOOT_TIME)
AttributeError: 'module' object has no attribute 'BOOT_TIME'
root@saturn /tmp/pcd8544/examples #
```
and
```
Traceback (most recent call last):
  File "./sys_info.py", line 78, in <module>
    main()
  File "./sys_info.py", line 75, in main
    stats()
  File "./sys_info.py", line 65, in stats
    lcd.text(mem_usage())
  File "./sys_info.py", line 45, in mem_usage
    usage = psutil.phymem_usage()
AttributeError: 'module' object has no attribute 'phymem_usage'
```
and
```
Traceback (most recent call last):
  File "./sys_info.py", line 79, in <module>
    main()
  File "./sys_info.py", line 76, in main
    stats()
  File "./sys_info.py", line 70, in stats
    lcd.text(network('wlan0'))
  File "./sys_info.py", line 57, in network
    stat = psutil.network_io_counters(pernic=True)[iface]
AttributeError: 'module' object has no attribute 'network_io_counters'
```

It needs to use
```
datetime.fromtimestamp(psutil.boot_time()
```

```
usage = psutil.virtual_memory()
```

```
    stat = psutil.net_io_counters(pernic=True)[iface]
```

It also doesn't work if there's no wlan0 interface. The RPi I'm running on only has eth0. You can use 

```
#!/usr/bin/python

import fcntl, socket, struct

def getHwAddr(ifname):
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    info = fcntl.ioctl(s.fileno(), 0x8927,  struct.pack('256s', ifname[:15]))
    return ''.join(['%02x:' % ord(char) for char in info[18:24]])[:-1]

try:
  print (getHwAddr('eth0'))
except:
  print ("eth0 interface not found")
try:
  print (getHwAddr('wlan0'))
except:
  print ("wlan0 interface not found")
```
To determine whether an interface exists.

DougieLawson, 2015-12-28:
Putting that all together I came up with this diff file

```
--- sys_info.py 2015-12-28 17:39:50.348930030 +0000
+++ sys_info.py~        2015-12-28 17:22:28.391519969 +0000
@@ -36,13 +36,13 @@

 def cpu_usage():
     # load average, uptime
-    uptime = datetime.now() - datetime.fromtimestamp(psutil.boot_time())
+    uptime = datetime.now() - datetime.fromtimestamp(psutil.BOOT_TIME)
     av1, av2, av3 = os.getloadavg()
     return "Load:%.1f %.1f %.1f Up: %s" \
             % (av1, av2, av3, str(uptime).split('.')[0])

 def mem_usage():
-    usage = psutil.virtual_memory()
+    usage = psutil.phymem_usage()
     return "Mem: %s %.0f%%" \
             % (bytes2human(usage.used), 100 - usage.percent)

@@ -53,7 +53,7 @@
             % (bytes2human(usage.used), usage.percent)

 def network(iface):
-    stat = psutil.net_io_counters(pernic=True)[iface]
+    stat = psutil.network_io_counters(pernic=True)[iface]
     return "%s:Tx%s,Rx%s" % \
            (iface, bytes2human(stat.bytes_sent), bytes2human(stat.bytes_recv))

@@ -66,7 +66,7 @@
     lcd.locate(0,2)
     lcd.text(disk_usage('/'))
     lcd.locate(0,3)
-    lcd.text(network('eth0'))
+    lcd.text(network('wlan0'))
     lcd.locate(0,4)

 def main():
```

rm-hull, 2015-12-28:
Oh nice one, I'll merge it in. This repo hasn't seen much action lately, so I guess the libs have moved on a bit

DougieLawson, 2015-12-29:
Beware: that diff file is upside down, because I'm an idiot (unlikely) or the diff util isn't intuitive (more likely).

closed 2015-12-29 by rm-hull

DougieLawson, 2015-12-30:
Thank you. I'll give that a go in the morning.

commit de0df0c6 (2015-12-29): Updated for latest psutil (patch from @DougieLawson) - fixes #9