rm-hull/luma.lcd #10: Not working on Raspberry Pi 2 model B

Pant, 2016-01-25:
I am trying to figure out why it is not working on me. Installation was fine but when I run any of the examples nothing happens.
Any help is appreciated!

MaartenB3, 2016-01-25:
Try changing the contrast, worked for me.

rm-hull, 2016-01-25:
Yes it is probably the same root cause as #6: some devices contrast are calibrated differently, so try setting the contrast when initialising.

The text examples work, but depending on the version of pillow installed the graphics demos may not. At the moment the ETA fix for that is unknown.

Pant, 2016-01-25:
Sorry for my noob question but how am I supposed to change the contrast?
Thank you for your answers!

rm-hull, 2016-01-25:
When you call init, call it with:

    lcd.init(contrast=67)

Permitted values 0..255.

You could also call the following anytime after initialising.

    lcd.set_contrast(183)

Pant, 2016-01-25:
I tryed the alphabet_text.py with the following ways: 

    if __name__ == "__main__":
    lcd.init(contrast=67)

and

    if __name__ == "__main__":
    lcd.init()
    lcd.set_contrast(183)

But neither works. Am i missing something? I Have triple-checked the wiring and i have used 2 lcds and non of the works.

rm-hull, 2016-01-25:
Can you drag/drop a closeup photo of your lcd module to this issue? There are a few different designs where the pinouts are different.

Also, does the backlight switch on when you power up?

Pant, 2016-01-25:
[image]
[image]
The led lights on when i run the alphabet_text.py. It does not lights off until next reboot!

Pant, 2016-01-26:
Is there anything else i could try?

rm-hull, 2016-01-26:
Ok, looking at your board, the pins are

* RST
* CE
* DC
* DIN
* CLK
* VCC
* LIGHT
* GND

This looks to be the same as mine :)

Can you annotate this list with the Raspberry Pi pins that you are connecting to? 
Presumably there are no dry solder joints or shorts?

There is some debugging output we can set - I'll dig out how to do this and paste it here.

Pant, 2016-01-26:
Guys sorry for the trouble but i figured out that the problem was that I used an external cord with 40 pins. Thank you for your assistance!

closed 2016-01-26 by Pant

rm-hull, 2016-01-27:
Ah, good - so you have it working properly now then?

Pant, 2016-02-04:
Yes everything works fine