rm-hull/luma.lcd #11: Initialization on PCD8544 board for raspberry

pevandenburie, 2016-02-13:
Hi there!
Thanks for your lib! Maybe you'll have some highlight on my initialization issue...

I bought the PCD8544 board which can be plug very easily on the Raspi, as you can see here:
https://github.com/pevandenburie/PCD8544/blob/master/PCD8544.pdf
The provided C demo (wiringPi based) works well.

However, I failed use your PCD8544 python wrapper...
To match the given layout:
CLK = GPIO17
D/C = GPIO27
CS = GPIO22
Din = GPIO18
RST = GPIO23

I tried following initialization:
LCDInit(CLK = 17, DIN = 18, DC = 27, RST = 23, LIGHT = 0, CE = 22);

I thought Light was just hardwired to 3.3V, and CE = CS.

But I got nothing on screen… Any idea on what I missed ?
Many thanks!

rm-hull, 2017-01-16:
I missed this issue completely - sorry for the non-reply.
Did you have any luck getting it working?
Please re-open if you still need assistance.

closed 2017-01-16 by rm-hull

pevandenburie, 2017-01-17:
No problem :-)
No I failed, my guess is all LCD' PINs are not wired on the "Raspberry PCD8544 board".
I wrote instead a NodeRED - Javascript wrapper over the C driver...

Thanks!

duchenpaul, 2018-12-04:
Hi @rm-hull 
I am having exactly the same issue as pevandenburie, could you please advise me how to configure the pcd8544, while those pins apart from RESET and D/C are totally different as the one specify in the doc, how can I redefine pins like SCLK, SDA, and CE?
Screen shield is the one he used: [PCD8544.pdf](https://github.com/pevandenburie/node-pcd8544-rpi/blob/master/PCD8544.pdf)

here is my pcd8544.conf file:
```
--display=pcd8544
--interface=spi
--spi-bus-speed=32000000
--gpio-reset=23
--gpio-data-command=27
--gpio-backlight=4
--width=84
--height=48
```

Thank you in advance, or any other one can help also thank you!