rm-hull/luma.lcd #12: no display, examples

zaidpirwani, 2016-06-03:
Hi,
I am trying different libraries, Adafruit one (https://learn.adafruit.com/nokia-5110-3310-lcd-python-library/overview) and yours.

when I run any example, the screen 
is blank and afterwards the Adafruit examples also show blank screen until I restart the screen - maybe something to do with the contrast setting, but am stuck.?

Please HELP, thanks.

programmin1, 2016-06-04:
Same here. Are you also using the plug-in 5110 like the one sold on Sainsmart? http://www.sainsmart.com/sainsmart-pcd8544-shield-ram-8448-mini-lcd-display-84-48-for-raspberry-pi-b-b.html

zaidpirwani, 2016-06-04:
nah, I made the circuit on my own. but it works with the Adafruit library so the wiring is okay and the wiring for this library is same - so i dont think this is a HW issue.

My Circuit so far:
[image]

rm-hull, 2016-06-08:
Hi, There are a number of things you should check:
* that the wiringPI library installed OK and all the other dependencies did too.
* double check the pinouts - I am aware that there are different layouts for the various PCD8544 modules out there.
* contrast is something someone else has suggested, but this can be changed easily in code, and the default is set quite high (i.e. quite dark)
* I have checked this library on a RPI B and the B2 model, but not the 3rd gen - if you are using a 3rd gen it is entirely possible that some low level h/w changed, but the version of the library doesnt support that (I can merge the latest changes from wiringPI to see if this is the case)

rm-hull, 2016-06-08:
Ok, I have merged the latest wiringPi code (see https://github.com/rm-hull/wiringPi/commit/ce2c58ac6654b8045c657f4cdc2028bfee3bc77c) - can you try pulling the wiringPi again, rebuilding it, and trying again?

programmin1, 2016-06-09:
Thanks, but I see the same behavior, no errors but nothing's happening running an example after install. 

BTW Looks like that circuit is the one in the documentation README? What do the buttons do? Does this library require that wiring because mine is just a PCB with LCD, that plugs in to the pi pins.

zaidpirwani, 2016-06-09:
nah the buttons arent needed for running the examples, they are just there for my project - I needed buttons and other sensors and RTC etc.

See this: http://github.com/zaidpirwani/piMon Currently using the Adafruit LCD Library - will try the changes later today and report back.

thijstriemstra, 2017-01-27:
A lot of changes have been made to this library. Please test with the latest and create a new issue if there's still a problem.

closed 2017-01-27 by thijstriemstra