rm-hull/luma.lcd #20: Examples not working for pcd8544

thijstriemstra, 2017-01-24:
I've wired up a pcd8544 device and connected it to an RPi2 with latest kernel and Python 3.5.3. When running an example, e.g. `python examples/3d_box.py -i spi --display pcd8544` it enables the backlight but doesn't show the actual 3d_box demo, it shows a line at the bottom that randomly appears on the screen. Other demo's don't show anything at all, only the backlight is enabled. Do all demo's work for you?

thijstriemstra, 2017-01-24:
Same random flashing bar for the bounce demo.

rm-hull, 2017-01-24:
Is the wiring as per the diagram in the docs? If so, create a file called **pcd8544.conf** with the content as follows:

```
--display=pcd8544
--interface=spi
--spi-bus-speed=32000000
--bcm-reset=24
--bcm-data-command=23
--bcm-backlight=18
--width=84
--height=48
```

Then invoke an example as follows:

```
$ examples/demo.py -f pcd8544.conf
```

The default values for the DC and RST pins in code are different to that on the wiring diagram

rm-hull, 2017-01-24:
Note to self: Should really check-offs some different configuration files for various devices

thijstriemstra, 2017-01-27:
Unfortunately that didn't work either, the screen stays on with active backlight but nothing happens. Attached some pictures. You sure that config is correct? Maybe my screen is broken..

[image]

[image]

rm-hull, 2017-01-27:
From just now:

```
$ pip list | grep luma
luma.core (0.1.13)
luma.lcd (0.2.1)
luma.led-matrix (0.4.2)
luma.oled (2.1.0)

$ cat pcd8544.conf 
--display=pcd8544
--interface=spi
--spi-bus-speed=32000000
--bcm-reset=24
--bcm-data-command=23
--bcm-backlight=18
--width=84
--height=48

$ python examples/greyscale.py -f pcd8544.conf
```

produces:

[image]

rm-hull, 2017-01-27:
Both the bounce and 3d box demos work as expected too

thijstriemstra, 2017-01-27:
Ok notice I have these pins soldered on the other side of the board. You're using the ones with an arrow, so let me start soldering ;)

rm-hull, 2017-01-27:
I hadnt noticed the arrow. at. all.

thijstriemstra, 2017-01-27:
I would expect the arrow side to be 'out' or something but I guess not.

rm-hull, 2017-01-27:
or maybe "this way up" 📦

thijstriemstra, 2017-01-27:
Unfortunately it still doesn't work, the line or bar that I saw before, is now visible on the bottom of the screen but other than that, empty screen. Can you post a picture of the back of your screen, I'll find a better camera and do the same for mine.

rm-hull, 2017-01-27:
Can't take a pic of the back unfortunately, the LCD is glued onto the stripboard with epoxy resin.

Did you try supplying 5V or 3V3 ? (try both)

Also try experimenting with setting different contrast ratios, some devices are calibrated differently

These are quite primitive controllers (compared to the SSD13xx at least), so there's not much to configure.

Alternatively, try a different lib (Adafruit one?) and see if you can get that working... also, i seem to remember: didnt you have some issues with one of your SPI oleds on a RPi3 ?

thijstriemstra, 2017-01-27:
This is on an RPi 2. How would i change the contrast ratio (that's probably it..)?

rm-hull, 2017-01-27:
The `demo.py` should cycle through the whole contrast range. If you run that does the image fade in & out?

https://github.com/rm-hull/luma.examples/blob/master/examples/demo.py#L60-L71

thijstriemstra, 2017-01-27:
No I don't see anything at all (at least 99% of the screen). I'll get my hands on a better camera tomorrow and posts some pictures.

thijstriemstra, 2017-01-27:
> or maybe "this way up"

well, the diagram seems to use the ones on the bottom (although there's not an arrow on the picture) judging from the rotation of the screen and the thicker side of the screen frame at the top of the picture.

This is why I initially soldered and wired on that side of the display.

https://luma-lcd.readthedocs.io/en/latest/_images/wiring-diagram.png

rm-hull, 2017-01-28:
I imagine the corresponding top and bottom pins are connected (e.g DIN to DIN, CLK to CLK, etc)

I mean "this way up" to just indicates the usual orientation, rather than anything to do with the pin connections

somu1795, 2017-03-22:
I think , it should be "--gpio-data-command" instead of "--bcm-data-command" , and "--gpio-reset" instead of "--bcm-reset" for the pins 23 and 24(which are gpio numbering).

thijstriemstra, 2017-03-22:
@somu1795 yep, you're right, it changed after this ticket was created.

closed 2021-11-23 by thijstriemstra