rm-hull/luma.lcd #28: RuntimeError('This module can only be run on a Raspberry Pi!')

thijstriemstra, 2017-03-02:
A `RuntimeError('This module can only be run on a Raspberry Pi!')` is thrown when running the examples on a non-raspberrypi. This should be a better luma.core.error.

rm-hull, 2017-03-02:
This is RPi.GPIO telling you off for not running on a BCM2835. 

You can swap the gpio module out for a `noop()` but this wont get you very far, ... so it would be nice to see if there were a more generic GPIO library, but when I last looked this wasn't the case, as it obviously relies on the low level chipset... although we could look at something that uses /sys/bus - that might involve writing a whole new layer, again something I had considered

thijstriemstra, 2017-03-02:
simply having a generic luma.core error would be fine with me.

thijstriemstra, 2017-03-08:
Should be fixed in next release.

closed 2017-03-09 by thijstriemstra