rm-hull/luma.lcd #35: Support for HT1621 Module Character LCD

thijstriemstra, 2017-03-21:
it's called `5V LCD Module 6Bit 8-Segment 3-wires SPI HT1621 Module Character LCD for Arduino`.

On ebay: http://www.ebay.com/itm/351939450953

thijstriemstra, 2017-03-21:
@rm-hull also sent one your way

rm-hull, 2017-03-21:
Awesome... 😆

rm-hull, 2017-04-11:
Arrived today, pulling together a driver now, rudimentaries are there:

[image]

It will most likely be driven like the max7219 by wrapping the `sevensegment class around the driver, eg.

```python

from luma.lcd.device import ht1621
from luma.core.virtual import sevensegment

device = sevensegment(ht1621())
device.text = "Hello"
```

_[the sevensegment class will need to be pulled into luma.core]_

thijstriemstra, 2017-04-11:
nice! expecting mine any day then.

rm-hull, 2017-04-11:
WIP on branch https://github.com/rm-hull/luma.lcd/tree/ht1621

Its getting there, but the segments are mapped differently to the max7219 segments, so not sure if it is a good idea to push those down into core. At the moment I just created a hack test script that converts the segment bytes on the fly.

[image]

thijstriemstra, 2017-04-11:
looks very crisp. but the battery icon is kinda burnt in there?

rm-hull, 2017-04-11:
Like any LCD, unless you view it directly head on, you see the slight ghosting of the unlit segments.
The viewing angles aren't great.
Also, can't appear to switch the backlight on or off (it's on all the time)

thijstriemstra, 2017-04-12:
Ok but its first screen i see with pre-defined icons as such.

PR #45 (2017-04-22): HT1621 driver
Fixes #35
closed 2017-04-22 by rm-hull