rm-hull/luma.lcd #39: ST7735S SPI tft

thijstriemstra, 2017-03-29:
I have this [1.44" inch Serial 128x128 SPI Full Color TFT LCD Display](http://www.ebay.com/itm/172097169256) with following specs:

- ST7735S

- SPI interface

Pins:

- 3V3
- GND
- SDI
- SCL
- CS
- RST
- RS
- BLK

but the docs at https://luma-lcd.readthedocs.io/en/latest/install.html#st7735 mention 16 pins.

Did I get an incompatible device because I'm confused how to hook it up.

rm-hull, 2017-03-29:
Ha! I got one as well, but haven't had the chance to connect it up yet. It should be compatible.
The pinouts detailed in the docs are for the device I already had which has a built in microSD card reader. 

Looking at the listing it should be straightforward-ish to map to the correct RPi pins. 

There will need to be a small fix to the st7735 device to accept width/height params (I took the decision, wrongly as it turns out, to hardcode to 160x128)

thijstriemstra, 2017-03-29:
One of my ssd13-something screens started smoking the other day so I'll wait for a confirmation ;)

rm-hull, 2017-03-29:
yikes!

closed 2017-04-07 by rm-hull

rm-hull, 2017-04-07:
@thijstriemstra 

Pinouts:

3V3 --> 3V3
GND --> GND
SDI --> [BCM 10](https://pinout.xyz/pinout/pin19_gpio10)
SCL --> [BCM 11](https://pinout.xyz/pinout/pin23_gpio11)
CS --> [BCM 8](https://pinout.xyz/pinout/pin24_gpio8)
RST --> [BCM 24](https://pinout.xyz/pinout/pin18_gpio24)
RS -> [BCM 23](https://pinout.xyz/pinout/pin16_gpio23)
BLK --> [BCM 18](https://pinout.xyz/pinout/pin12_gpio18)

thijstriemstra, 2017-04-08:
awesome, thanks

rm-hull, 2017-04-08:
@thijstriemstra there should be a `config/st7735_128x128.conf` in the examples repo which is suitable.

PR #41 (2017-04-07): Add support for 128x128 version of ST7735 display
Fixes #39 and #40
commit 116c09ab (2017-04-07): Add support for 128x128 version of ST7735 display (#41)

Fixes #39 and #40

* Add support for 128x128 version of ST7735 display
* Add H/V offsets
* Make backlight configurable active high/low