rm-hull/luma.lcd #40: 1.44 TFT LCD ST7735S is not working as expected

mnrozhkov, 2017-04-03:
Hi! I have this [1.44" SPI 128x128  TFT LCD Display (RED PCB V1.1)](https://github.com/sumotoy/TFT_ILI9163C/wiki) with following specs:
conf/st7735.conf: 
```
--display=st7735
--interface=spi
--spi-bus-speed=16000000
--gpio-reset=24
--gpio-data-command=23
--gpio-backlight=18
--width=128
--height=128
```

1) When running an example  `sudo python examples/3d_box.py -f conf/st7735.conf` , there is nothing on the screen! Event the backlight is not enabled. 

2) After I connect `gpio-backlight` pin directly to 3.3V source - it works fine 

Notes: 
1. It seems that code can't reach/control some of the required GPIO or numbering is wrong. (I've used BCM numbering)  

2. Even when I follow ['Python usage' instructions](https://luma-lcd.readthedocs.io/en/latest/python-usage.html#) nothing happens when I try 'draw.rectangle' or 'draw.text'

3. When I try to get the 'device' object details in runtime (run with -f conf/st7735.conf), I see the unexpected arguments for screen size (my LCD has width=128, height=128): 
```
luma.lcd.device.st7735 at 0x765da3b0 (
  bounding_box = (0, 0, 159, 127),
  framebuffer = luma.core.framebuffer.diff_to_previous at 0x765da7f0 (
      bounding_box = (0, 0, 160, 128),
      image = PIL.Image.Image at 0x76359b90 (
          category = 0,
          format = None,
          format_description = None,
          height = 128,
          im = ImagingCore at 0x76d6afa0 (
              bands = 3,
              id = 22136808,
              mode = 'RGB',
              ptr = PyCapsule at 0x7655f020 (),
              size = (160, 128),
              unsafe_ptrs = ((...), (...), (...), (...), (...), ..., (...), (...), (...), (...), (...))
            ),
          ...,
          palette = None,
          pyaccess = None,
          readonly = 0,
          size = (160, 128),
          width = 160
        )
    ),
  height = 128,
  mode = 'RGB',
  rotate = 0,
  size = (160, 128),
  width = 160
)
```

Do you have any suggestions?

rm-hull, 2017-04-04:
See the comment in #39 - i took a decision (wrongly) to hardcode the resolution to 160x128 thinking that was the only resolution. I will push a change so it supports 128x128 but it won't be for another few days.

As to the issue with the backlight, i don't know. I have bought one of these devices but haven't had the chance to plug it in. I will update the ticket with my findings

rm-hull, 2017-04-07:
@mnrozhkov there is a new version that supports 128x128 variants of this device. If you upgrade with:

    $ sudo -H pip install --upgrade luma.core luma.lcd

The backlight was originally written for the PCD8544 which is enabled as active-low, whereas the ST7735 enables the backlight as active-high. It is now configurable, so supports either setting.

Lastly, if you then pull from master for the luma.examples repo, and you should find a `conf/st7735_128x128.conf` file with the correct settings (well, this works on the 128x128 device I have).

Let me know how you get on,

thijstriemstra, 2017-04-11:
@mnrozhkov any news?

mnrozhkov, 2017-04-11:
@rm-hull thanks a lot! It works well. I've updated library and changed conf.  

I use LCD to build eyes for my robot. I'll share with you how it will word as soon as I finish this. Your library is priceless help for our project! 

Best wishes!

closed 2017-04-11 by mnrozhkov