rm-hull/luma.lcd #42: 1.8 Inch SPI TFT LCD Module Display ST7735B

thijstriemstra, 2017-04-11:
- Ebay: http://www.ebay.com/itm/401010353521

[image]

It has the following specs but I'd like to know for sure how to hook it up:

1.8\" Serial Port SPI TFT LCD Display Module
Size: 1.8 inch
Dot Matrix: 128*160
Module Size: 44.96(length)X53.47mm(width)
Display Area Size: 35.04(length)*28.03mm(width)
Input Voltage: 5V/3.3V
Pin Definition: 1-RST 2-CE 3-D/C 4-DIN 5-CLK 6-UCC 7-BL 8-GND
It has PCB backplane with power IC, SD card socket
It need 4 IO port at least to drive.
The module port is compatible with 1602 LCD and Nokia 5110/3310 LCD Display port.
package included:1pc 1.8\" SPI TFT LCD Display Module Serial 128*160 PCB Adapter Power IC SD Socket

rm-hull, 2017-04-11:
@thijstriemstra 

Pinouts:

RST --> [BCM 24](https://pinout.xyz/pinout/pin18_gpio24)
CS --> [BCM 8](https://pinout.xyz/pinout/pin24_gpio8)
D/C --> [BCM 23](https://pinout.xyz/pinout/pin16_gpio23)
DIN --> [BCM 10](https://pinout.xyz/pinout/pin19_gpio10)
CLK --> [BCM 11](https://pinout.xyz/pinout/pin23_gpio11)
VCC --> 3V3 or 5V0 (dependent on whether J1 is shorted or open)
BL --> [BCM 18](https://pinout.xyz/pinout/pin12_gpio18)
GND --> GND

The st7735.conf should work, but you may need to tweak the spi-bus-speed down to 16000000 or 8000000 here https://github.com/rm-hull/luma.examples/blob/master/conf/st7735.conf#L3

thijstriemstra, 2017-04-12:
Thanks, got it working!

closed 2017-04-12 by thijstriemstra