rm-hull/luma.lcd #46: Add support for 128x64 ST7920 LCD

rm-hull, 2017-04-23:
* Ebay listing: http://www.ebay.co.uk/itm/302089090235
* Datasheet: [ST7920.pdf](https://github.com/rm-hull/luma.lcd/files/950110/ST7920.pdf)
* Ref: http://www.filipeflop.com/pd-6b7ea-display-lcd-grafico-128x64.html

[image]

rm-hull, 2017-06-12:
## Implementation notes

Code [WIP] on branch: [feature/st7920](https://github.com/rm-hull/luma.lcd/tree/feature/st7920).
Documentation preview: http://luma-lcd.readthedocs.io/en/feature-st7920/

[image]

thijstriemstra, 2018-02-04:
@rm-hull can you open a PR for that branch so people could start testing/take a look?

mlatorre31, 2019-07-07:
Hi,

I have a ST7920 LCD. If you need any testing, let me know as there is no python library with i2c for this device

X-Stas-EEE, 2019-09-29:
Is there any news about the st7920 support?