rm-hull/luma.lcd #61: Add support for Sharp memory display

rm-hull, 2017-12-14:
Implementation Notes
----------------------------

* Available from: https://www.adafruit.com/product/1393
* Datasheet: https://www.sharpsma.com/documents/1468207/1485747/Memory+LCD+Theory%2C+Programming%2C+and+Interfaces
* App note: https://cdn.sharpsde.com/fileadmin/products/Displays/2016_SDE_App_Note_for_Memory_LCD_programming_V1.3.pdf
* Linked (original) issue: https://github.com/rm-hull/luma.core/issues/113
* Code [WIP] on branch: feature/sharp-lcd.
* Documentation preview: http://luma-lcd.readthedocs.io/en/feature-sharp-lcd/

[image]

[image]

rm-hull, 2018-01-02:
References: http://www.makerdyne.com/blog/category/electronics/breakout-boards/

CRImier, 2023-12-16:
I'm interested in adding support for Sharp Memory displays to luma.lcd, currently looking at the #63 code. Would such contribution be interesting?

thijstriemstra, 2023-12-16:
> Would such a contribution be interesting to upstream?

I think so!