rm-hull/luma.lcd #62: Pixel artifacts on ST7735

sgjava, 2017-12-25:
I'm using this https://www.amazon.com/gp/product/B01J76ZGMY ST7725 on my ODROID C1+ which I have SSD1306 and SSD1331 working fine. When I use the examples or a sample program I wrote it leaves pixel artifacts on the bottom and right side of the display. See attached image:

[image]

thijstriemstra, 2017-12-25:
Can you show us the example code to doublecheck for errors?

sgjava, 2017-12-25:
Sure for the clock.py I used:

`sudo python clock.py -d st7735 -i spi --spi-port 0 --spi-device 0 --gpio-data-command 24 --gpio-reset 25 --gpio-backlight 13 --width 160 --height 128 --backlight-active high`

My little sample:

```
import time
from RPi import GPIO
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.lcd.device import st7735

GPIO.setmode(GPIO.BOARD)
GPIO.setwarnings(False)
serial = spi(port=0, device=0, gpio=GPIO, gpio_DC=18, gpio_RST=22)
device = st7735(serial)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white")
    draw.text((3, 3), "Hello", fill="white")

time.sleep(10)
```

Both result in the exact same artifacts.

thijstriemstra, 2017-12-25:
what if you use 161x31 or 159 x 29? just a thought

sgjava, 2017-12-25:
Anything except 160x128 results in **Unsupported display mode: 159 x 127**

sgjava, 2017-12-25:
OK, I tried another little sample piece of code and it looks like the canvas is shifted as the following displays a bounding box (with 2 rows of pixel clutter at the bottom):

```
with canvas(device) as draw:
    draw.line((0, 2, 159, 2), fill="white")
    draw.line((1, 0, 1, 127), fill="white")
    draw.line((0, 127, 159, 127), fill="white")
    draw.line((159, 0, 159, 127), fill="white")
time.sleep(5)
```

rm-hull, 2017-12-25:
So the different manufacturers map the screen to different memory offsets. I had to add `h-offset` and `v-offset` for the ST7725 ... Perhaps we need to do the same for this device too.

See https://github.com/rm-hull/luma.core/blob/master/luma/core/cmdline.py#L238 and https://github.com/rm-hull/luma.lcd/blob/master/luma/lcd/device.py#L146

rm-hull, 2017-12-25:
Sorry not paying attention properly, thought this was about ssd1351... Try fiddling with the --h-offset and --v-offset params either +/-1 or 2 until it displays properly

rm-hull, 2017-12-25:
The size should be 160x128. Use the examples demo.py script as a reference to get the offsets correct

sgjava, 2017-12-25:
I'll try that tomorrow and check back with the results.

thijstriemstra, 2017-12-25:
ah yeah... you can force it by removing this check though: https://github.com/rm-hull/luma.lcd/blob/master/luma/lcd/device.py#L154

But it's just a wild guess, I don't expect it to improve.

sgjava, 2017-12-26:
--h-offset 1 --v-offset 2 worked perfect!

`sudo python clock.py -d st7735 -ipi --spi-port 0 --spi-device 0 --gpio-data-command 24 --gpio-reset 25 --gpio-backlight 13 --width 160 --height 128 --backlight-active high --h-offset 1 --v-offset 2

closed 2017-12-26 by sgjava