rm-hull/luma.lcd #70: Examples not working for waveshare 1,44inch lcd hat (128x128, 65k)

LocutusOfPenguin, 2018-08-10:
Hi,
running the examples produces strange output on the display. Strange = something like it seems to generate the picture all over again, but the pict are some sort of nonsense-pixels. Hard to explain.

I tried to use the ST7735 display driver  or the conf file for it ... no success.

Using the files from waveshare works, but i would like to use this luma :-)
Also other displays like SH1306 works with luma for me. So only this one is not working.

Please help.
Jürgen

thijstriemstra, 2018-08-10:
Can you post picture and config file?

LocutusOfPenguin, 2018-08-10:
If i knew how to do it i would do (lateron), haha - i use Github for long, but never needed to add a pict ...Sorry.

The conf file, is what is inside the examples...not changed..But here y are:

--display=st7735
--interface=spi
--spi-bus-speed=16000000
--gpio-reset=24
--gpio-data-command=23
--gpio-backlight=18
--width=160
--height=128
--backlight-active=high

LocutusOfPenguin, 2018-08-10:
ups....i also use conf with the -width parameter = 128 (so both!) ...perhaps thats the problem?
Im not with this hardware right now, so i cant check (now)

LocutusOfPenguin, 2018-08-10:
sudo python3 ./colors.py  -d st7735 --height 128 -i spi
Display: st7735
Interface: spi
Dimensions: 128 x 128
----------------------------------------

Display is flashing white & black (each 100%)

if i use "bounce.py" its flashing alot quicker

If i use the "--config", it stays white right now (nothing happening).
Its abit different to last time, i tried.

wifi4u, 2019-02-16:
@LocutusOfPeng, I have a 1.8 inch Waveshare ST7735, which has 160x128 pixels. You might try this config, which works for me (well, most of the time):

```--display=st7735
--interface=spi
--spi-bus-speed=16000000
--gpio-reset=24
--gpio-data-command=23
--gpio-backlight=18
--width=160
--height=128
--backlight-active=high```

However, the display's backlight always stays on if GPIO18 is not at low level. No sure whether this is intented, but the GPIO port falls back to mode `IN` if no script is running, which causes the backlight to be turned on again after it was turned off shortly. This might be caused by an internal pull-up resistor in the display's controller, one can see that brightness is slightly increased if BL input is connected to Vcc.

wifi4u, 2019-02-16:
@LocutusOfPeng, I have a 1.8 inch Waveshare ST7735, which has 160x128 pixels. You might try this config, which works for me (well, most of the time):

    --interface=spi
    --spi-bus-speed=16000000
    --gpio-reset=24
    --gpio-data-command=23
    --gpio-backlight=18
    --width=160
    --height=128
    --backlight-active=high

However, the display's backlight always stays on if GPIO18 is not at low level. No sure whether this is intented, but the GPIO port falls back to mode `IN` if no script is running, which causes the backlight to be turned on again after it was turned off shortly. This might be caused by an internal pull-up resistor in the display's controller, one can see that brightness is slightly increased if BL input is connected to Vcc.

wifi4u, 2019-02-16:
@LocutusOfPeng, I have a 1.8 inch Waveshare ST7735, which has 160x128 pixels. If your display is the 1.8 inch, too, you might try this config, which works for me (well, most of the time, sometimes the demo doesn't display anything, restarting the demo then helps):

--interface=spi
--spi-bus-speed=16000000
--gpio-reset=24
--gpio-data-command=23
--gpio-backlight=18
--width=160
--height=128
--backlight-active=high
However, the display's backlight always stays on if GPIO18 is not at low level. No sure whether this is intented, but the GPIO port falls back to mode IN if no script is running, which causes the backlight to be turned on again after it was turned off shortly. This might be caused by an internal pull-up resistor in the display's controller, one can see that brightness is slightly increased if BL input is connected to Vcc.

wifi4u, 2019-02-16:
@LocutusOfPeng, I have a 1.8 inch Waveshare ST7735, which has 160x128 pixels. If your display is the 1.8 inch too, you might try this config, which works for me (well, most of the time, sometimes the demo doesn't display anything, restarting the demo then helps):

    --interface=spi
    --spi-bus-speed=16000000
    --gpio-reset=24
    --gpio-data-command=23
    --gpio-backlight=18
    --width=160
    --height=128
    --backlight-active=high

However, the display's backlight always stays on if GPIO18 is not at low level. No sure whether this is intented, but the GPIO port falls back to mode IN if no script is running, which causes the backlight to be turned on again after it was turned off shortly. This might be caused by an internal pull-up resistor in the display's controller, one can see that brightness is slightly increased if BL input is connected to Vcc.

LocutusOfPenguin, 2019-02-19:
@wifi4u thanks! 
My name missing a "uin" in yr text...its useless from github to shorten 3 chars with 3 other chars, ha.

No, i use a 128x128 ...but i give it a try .... just now im very busy (and this display is no longer my main focus, right now - needed to wait too long).

Rippert, 2019-03-26:
Just got my Wavshare 1.44 inch hat yesterday, and had the same problem. Went to the Waveshare wiki and looked up the schematic. It's on there(www.waveshare.com/wiki/1.44inch_LCD_HAT), and I was able to modify the st7725_128x128.conf file with the correct pinout:

```
--display=st7735
--interface=spi
--spi-bus-speed=16000000
--gpio-reset=27
--gpio-data-command=25
--gpio-backlight=24
--width=128
--height=128
--bgr
--h-offset=1
--v-offset=2
--backlight-active=high
```
All the luma.examples work fine with it using this configuration file. Maybe it could be added to the other stock configurations as `st7735_128x128_WShat.conf` or some such.

LocutusOfPenguin, 2019-07-25:
yes, it works now very well. THANKS!

Sorry, for long delay.

closed 2019-07-25 by LocutusOfPenguin