rm-hull/luma.lcd #78: Please advise a way to specify pin apart from D/C and RST

duchenpaul, 2018-12-05:
Maybe duplicate with #11, but I think I need a clear answer.
I got the same LCD as what described in #11 , so I have to reassign the PIN in the code.
I went over the doc, only found how to change D/C and RST, but no description of the rest ones (SCLK, SDA, and CE)
How do I change them?
Thank you

[Link to LCD I got](https://github.com/pevandenburie/node-pcd8544-rpi/raw/master/PCD8544.pdf)
[More detail](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=180569)

rm-hull, 2018-12-05:
By default PCD8544 driver is SPI compatible, hence the reason it is fixed to use the RPIs SPI interface pins.. this allows us to use the SPI hardware and the Linux kernel to drive the device efficiently. If you have bought a board which interfaces those lines to non-SPI griot header pins, then that is a poor decision by the board designer!

The only choice you have really is therefore to do what is called bit-banging in software .. this is what 
https://github.com/pevandenburie/node-pcd8544-rpi/blob/master/src/PCD8544.cc#L812 does.

Luma.lcd will support bitbanging but I haven’t tried it for a while. Instead of SPI, you would need to look at using this serial interface instead: 
https://github.com/rm-hull/luma.core/blob/master/luma/core/interface/serial.py#L118

rm-hull, 2018-12-05:
To be honest I would just source a different LCD board from ebay or aliexpress ... one where you wire it yourself

duchenpaul, 2018-12-06:
Thank you for your quick reply! 😄 
I tried bitbang in your core before, seems not working.
I will take another look when I got time

[image]

duchenpaul, 2018-12-06:
Hi @rm-hull
Hope it is something wrong on my side, I used below code, but no luck, nothing shows on my screen

please checkout [Front pic](https://github.com/duchenpaul/sys_status/raw/master/src/demo.png) to see what front looks like.

```python
from luma.core.interface.serial import spi, bitbang
from luma.core.render import canvas
from luma.lcd.device import pcd8544

# serial = spi(port=0, device=1, gpio_DC=27, gpio_RST=23)
serial = bitbang(SCLK=17, SDA=18, CE=22 , DC=27 ,RST=23)
device = pcd8544(serial, rotate=1)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
    draw.text((30, 40), "Hello World", fill="red")

```

Now that I won't use stock hardware SPI, I think below operation is meaningless, but I want to show you still
```
[ 22:22 - 192.168.2.107  ]
pi@Saturn ~/projects/PCD_8544_screen/python/test $ ls -l /dev/spi* 
crw-rw---- 1 root spi 153, 0 Dec  5 23:19 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Dec  5 23:19 /dev/spidev0.1
[ 22:23 - 192.168.2.107  ]
pi@Saturn ~/projects/PCD_8544_screen/python/test $ lsmod
Module                  Size  Used by
cmac                    3239  1
bnep                   12051  2
hci_uart               20020  1
btbcm                   7916  1 hci_uart
bluetooth             365780  24 hci_uart,bnep,btbcm
spidev                  7373  0
brcmfmac              292632  0
brcmutil                9863  1 brcmfmac
cfg80211              544545  1 brcmfmac
rfkill                 20851  6 bluetooth,cfg80211
snd_bcm2835            24427  0
snd_pcm                98501  1 snd_bcm2835
snd_timer              23968  1 snd_pcm
snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0
spi_bcm2835             7596  0
uio_pdrv_genirq         3923  0
fixed                   3285  0
uio                    10204  1 uio_pdrv_genirq
i2c_dev                 6913  0
ip_tables              13161  0
x_tables               20578  1 ip_tables
ipv6                  408900  43
[ 22:23 - 192.168.2.107  ]
pi@Saturn ~/projects/PCD_8544_screen/python/test $ 
```

thijstriemstra, 2021-11-23:
@duchenpaul got it to work?

duchenpaul, 2021-11-23:
Yes, I will close it.

closed 2021-11-23 by duchenpaul