rm-hull/luma.lcd #82: LCD ST7735 128X128  - WHITE SCREEN

gigabauertech, 2019-01-27:
Hello guys how are you all doing? Its my first time ever using GitHub so I'm sorry if I'm not using the right place to find an answer to my problem. 

#### Type of Raspberry Pi and LCD specifications
Raspberry Pi 3B+
ST7735 LCD SPI 1.44 128x128

Place where I bought the LCD:
[https://www.aliexpress.com/item/GREATZT-1PCS-5V-3-3V-1-44-inch-TFT-LCD-Display-Module-128-128-Color-Sreen/32948378014.html?spm=a2g0s.9042311.0.0.1f614c4dALAgbo](url)

#### Linux Kernel version
Linux retropie 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
Library: luma.lcd 1.1.0 (luma.core 1.8.3)
Python-dev: 2.7.13-2
Python-pip: 9.0.1-2+rpt2

#### Expected behaviour
I am willing to use this LCD to show at full time all the status of my Raspberry. Such as the cpu temp, storage, IP address and etc. My plan is to make this script start up as soon as my rasp starts too. So I created a sh file to be triggered by the rc.local all the time it boots up. 

My gpio set up is just like the library document:
led 18
sck 11(sclk)
sda 10(mosi)
ao 23
reset 24
cs 8(ceo)
gnd
vcc 3.3/5

This is how it looks whenever it works:
[image]

This is my status script: https://paste.ofcode.org/LuY4CDcwSmMcXjGzUBjXx

This is the sh script witch the rc.local triggers when the rasp boots up:
[image]

#### Actual behaviour
My actual problem is that sometimes the LCD will turn on but completely white. Even if you check it closer you won't be able to see anything or even blinking. 

[image]

So, to make it work I go to the command line and and input:

[image]

Even running this command in the manual way the screen will shows itself all white, so I have to interrupt and run again until it shows the status screen as I showed before.

I tried to play with those commands from the library such as the device.clean, decive.show however I will randomly get this white screen. 

Have you ever seen this type of issue? 

Note: I tried to change from the 3.3 to 5.0 but the problem is still the same.
Note_2: I also tried to wait a while to see if it would change but nothing happens the white screen will stay up until you interrupt.

In case I was not clear please let me know so I can even make a video. Thanks for this amazing library and also for the attention.

LocutusOfPenguin, 2019-02-19:
Hi,
perhaps similar problem as my #70 ?
Your status script url doesnt work.

Rippert, 2019-03-27:
If it works intermittently, then I would suspect a hardware problem. You should check your jumper wires carefully, with a multimeter if you have one. Other than that, I would suspect that the LCD breakout board has some kind of problem. Look on the back for bad solder joints on the discrete elements and check the ribbon cable coming out of the side of the display for damage (may need a magnifying glass). The reviews I've seen of this particular board indicate a high rate of failure so maybe you just got a bad one.

It's possible that your RPi GPIO interface has problems also, though less likely. You could alter the .conf file to change the pins it calls out to other ones and rewire to see if that helps. You can only do this for the Reset, DC (A0) and Backlight (LED) pins, but those could be the problem.

gigabauertech, 2019-04-16:
@LocutusOfPenguin

Sorry for the long reply, and my issue it's just a bit similar. The white screen shows in random times. It is always white never black.

@Rirppert

Thanks for the advice! I will try it today and update you guys, and also update the scripts link.

LouisPi, 2019-09-29:
Any luck with fixing this?

thijstriemstra, 2020-01-19:
another report about a white screen: #93

jakiee3y, 2020-03-04:
some delay should be add to hw-reset and sw-reset.
in luma/core/interface/serial.py
add 0.1s delay before and after "self._gpio.output(self._RST, self._gpio.HIGH) "
in luma/lcd/device.py
add 0.1s delay before and after "self.command(0x11)                      # sleep out & booster on"
maybe the delay can be lower, but the four 0.1s work for me.

rm-hull, 2020-03-04:
@jakiee3y thanks for the heads up ... I will add that & hopefully this will resolve these issues

jakiee3y, 2020-03-30:
> some delay should be add to hw-reset and sw-reset.
> in luma/core/interface/serial.py
> add 0.1s delay before and after "self._gpio.output(self._RST, self._gpio.HIGH) "
> in luma/lcd/device.py
> add 0.1s [...]

If somebody hits here, you can try to re_init st7735 by following function without changing lib code.

import time
def init_st7735(device):
    # self._serial_interface.
    #    self._gpio.output(self._RST, self._gpio.LOW)  # Reset device
    #    self._gpio.output(self._RST, self._gpio.HIGH)  # Keep RESET pulled high
    # RGB or BGR order
    bgr = False
    order = 0x08 if bgr else 0x00
    spi = device._serial_interface
    spi._gpio.output(spi._RST, spi._gpio.LOW)
    time.sleep(0.1)
    spi._gpio.output(spi._RST, spi._gpio.HIGH)
    time.sleep(0.1)
    device.command(0x01)                      # reset
    time.sleep(0.1)
    device.command(0x11)                      # sleep out & booster on
    device.command(0xB1, 0x01, 0x2C, 0x2D)    # frame rate control: normal mode
    device.command(0xB2, 0x01, 0x2C, 0x2D)    # frame rate control: idle mode
    device.command(0xB3, 0x01, 0x2C, 0x2D,    # frame rate control: partial mode dot inversion mode
                 0x01, 0x2C, 0x2D)          # frame rate control: line inversion mode
    device.command(0xB4, 0x07)                # display inversion: none
    device.command(0xC0, 0xA2, 0x02, 0x84)    # power control 1: -4.6V auto mode
    device.command(0xC1, 0xC5)                # power control 2: VGH
    device.command(0xC2, 0x0A, 0x00)          # power control 3: OpAmp current small, boost freq
    device.command(0xC3, 0x8A, 0x2A)          # power control 4: BCLK/2, Opamp current small & Medium low
    device.command(0xC4, 0x8A, 0xEE)          # power control 5: partial mode/full-color
    device.command(0xC5, 0x0E)                # VCOM Control 1
    device.command(0x36, 0x60 | order)        # memory data access control
    device.command(0x20)                      # display inversion off - another value is 0x21
    device.command(0x3A, 0x06)                # interface pixel format
    device.command(0x13)                      # partial off (normal)
    device.command(0xE0,                      # gamma adjustment (+ polarity)
                 0x0F, 0x1A, 0x0F, 0x18, 0x2F, 0x28, 0x20, 0x22,
                 0x1F, 0x1B, 0x23, 0x37, 0x00, 0x07, 0x02, 0x10)
    device.command(0xE1,                      # gamma adjustment (- polarity)
                 0x0F, 0x1B, 0x0F, 0x17, 0x33, 0x2C, 0x29, 0x2E,
                 0x30, 0x30, 0x39, 0x3F, 0x00, 0x07, 0x03, 0x10)

    device.clear()
    device.show()

thijstriemstra, 2020-03-30:
@jakiee3y can you make a pull request with your fix(es)?

jakiee3y, 2020-03-30:
The delay adding to luma/core/interface/serial.py will effect all devices using spi.

thijstriemstra, 2020-03-30:
ah, nevermind, that'll need a different solution then.

LouisPi, 2020-03-30:
> If somebody hits here, you can try to re_init st7735 by following function without changing lib code.
>
> ```
> import time
> def init_st7735(device):
>     # self._serial_interface.
>     # [...]

Thanks for the heads up about this fix. I haven't looked at my project using that used Luma LCD but now's the perfect time to take a look again. Where do I put this code?

thijstriemstra, 2020-03-30:
@LouisPi can you test pull request #105?

jakiee3y, 2020-03-31:
pull request #175 to luma.core makes the addtional flags (timed_reset, inverse) available to cmdline .

jakiee3y, 2020-03-31:
> Thanks for the heads up about this fix. I haven't looked at my project using that used Luma LCD but now's the perfect time to take a look again. Where do I put this code?

init_st7735 should be called after get_device() in luma.examples.
like this: (3d_box.py)
`from init_st7735 import init_st7735
if __name__ == "__main__":
    try:
        device = get_device()
        init_st7735(device)
        main()
    except KeyboardInterrupt:
        pass
`

rm-hull, 2020-09-20:
This will be fixed by implementing a delay when resetting the interface in luma.core (PR to follow)

PR #105 (2020-09-20): Support inverted display on ST7735
try to init the odd version of st7735

~~fixes #82~~
fixes #105