rm-hull/luma.lcd #84: aux.py can't write on MS Windows

pasquyonline, 2019-04-24:
Hi, guys,
first, this library is wonderfull, great job! second, I need to clone the project on MS Windows machine, but on this operating system (any version!!!) a file can't be named "**aux.xx**" (_xx_ stand for anything you want) so the file **aux.py** can't be created.

So, my request is: it is possible to rename this file to **_aux.py** for example?

thijstriemstra, 2019-04-24:
we're talking about https://github.com/rm-hull/luma.lcd/blob/master/luma/lcd/aux.py right? Are you serious, this would blow my mind..

pasquyonline, 2019-04-24:
Believe me, I'm very serious! I didn't even believe it, then I did tests on Windows 10 and Windows 7: same error message.

This is a link that describe "the thing"!!
https://www.reddit.com/r/linuxmasterrace/comments/9ukft0/windows_10_still_cant_create_a_file_called_auxtxt/?depth=2

pasquyonline, 2019-04-24:
I tried also on a Windows Xp on a virtual machine and also in that operating system there is that limitation!

rm-hull, 2019-05-26:
Related issue: #44

thijstriemstra, 2019-05-26:
@rm-hull what are your thoughts on renaming it?

rm-hull, 2019-05-26:
Wondered about creating a `backlit_device` class with this functionality, and then those devices that have a backlight would inherit from that instead of `device`, and the we could just remove the aux.py file completely.

thijstriemstra, 2019-05-26:
sounds good. removing a module is backwards-incompatible though (so bump version to nxt level?)

rm-hull, 2019-05-26:
> so bump version to nxt level?

Agreed - it would have to be v2.0.0

PR #86 (2019-06-02): backlit device class
Fixes #84, #44

- [x] Working implementation
- [x] Documentation
- [x] Regression tests
- [x] `luma.core` cmd-line arg changes
- [x] Needs merging: https://github.com/rm-hull/luma.core/pull/163
- [x] Update setup.py to depend on specific version of luma.core with above PR
- [x] Add breaking changes notice on README
commit cc2db4d9 (2019-06-02): Backlit device class (#86)

Fixes #84, #44

* Backlit device class
* Fix tests
* Switch backlight off on clean-up
* Remove aux.py file
* Repurpose tests
* Fix API documentation
* Additional test & persist functionality
* Add version
* Assume GPIO setmode is called elsewhere in luma.core (or outside when unmanaged)
* Convert remaining display devices to inherit from backlit_device
* Update docs, illustrating new device.backlight() usage
* remove serial interface - not needed for ht1621
* Require luma.core >= 1.11
* Update docs
closed 2019-06-02 by rm-hull