rm-hull/luma.lcd #89: ST7735 FPS

rollalt, 2019-06-23:
Raspberry pi 2B
Linux raspberrypi 4.19.42-v7+ #1219 SMP

Is there an option to modify the frame rate?

rm-hull, 2019-06-24:
The only things that you can realistically do are:
* Try increasing the SPI bus speed parameter - some devices/systems and displays can tolerate 32MHz
* Get a faster RPi (I hear the 4B is launched today)
* I cant remember off the top of my head whether we implemented the diff-to-previous framebuffer for the ST7735, but if we did, it will only draw the region that changed since the last frame draw - this can significantly speed things up.

I have an idea bubbling to use native linux framebuffers (see https://github.com/rm-hull/luma.lcd/issues/67#issuecomment-500174784), but haven't gotten around to doing any work on that yet - that should improve performance significantly when implemented

rollalt, 2019-06-30:
@rm-hull can you explain please how to use diff-to-previous framebuffer for the ST7735, or how I can check if it is implemented?
Thanks.

closed 2021-11-23 by thijstriemstra