rm-hull/luma.lcd #93: ST7735 White Screen

LouisPi, 2019-09-29:
Any ideas what could be causing this? Is what working when I last used it a few months ago.

thijstriemstra, 2019-09-30:
can you include more info: os, python, example script etc?

LouisPi, 2019-10-01:
I'm using Raspian (Stretch I think). I've tried the Luma example scripts using the config for my LCD, as well as the script for my project. I've checked pins and wires (replaced all wires) and it turned out there was a problem with my audio board and the LCD sharing the pin the LCD uses for the backlight but after fixing this, the screen was still white. The audio board wasn't actually in use, and I believe the backlight has worked throughout (hence the white screen) so I don't think this has anything to do with the issue. 

I can't see what software could be at fault, considering my LCD worked before. Maybe it's worth trying the LCD on a different pi.

Thanks,
Louis

thijstriemstra, 2020-03-30:
Looks like a duplicate of #82. Can you give the suggested changes there a try @LouisPi 

Marking as duplicate.

closed 2020-03-30 by thijstriemstra