rm-hull/luma.lcd #97: How to add more types of screens ?

markbirss, 2019-12-26:
How do i add for example the ST7920 oled screen ?

rm-hull, 2019-12-26:
Moving to luma.lcd ....

markbirss, 2019-12-26:
Ok, understood, thank you for your feedback

I will close the issue but leave these notes below

I have found much of that information but,  I dont have the depth of understanding at the moment to put it into a implementation.

Will leave it here if anyone else interested.

Datasheet
https://www.lcd-module.de/eng/pdf/zubehoer/st7920_chinese.pdf

Enable SPI mode is simply done by connecting PSB to GND
Enable Parallel mode by connecting PSB to 5V

How th screen works
https://www.instructables.com/id/The-Secrets-of-an-Inexpensive-Ubiquitous-Chinese-L/

U8G2 ports 
https://github.com/ribasco/u8g2-rpi-demo

https://github.com/olikraus/u8g2

closed 2019-12-26 by markbirss

rm-hull, 2019-12-26:
Related issue: #46

rm-hull, 2019-12-26:
I think the PR for this stalled because driving it via parallel bit-banging was going to be awkward. 
If this is true:

> Enable SPI mode is simply done by connecting PSB to GND

then it might be straightforward to getting it working