rm-hull/luma.lcd #101: Define the screen rotation inside of the program

raf59, 2020-01-19:
I use a st7735 device on a RPI3b.
It's working if I use the get_device() function from demo_opts.py as in luma.examples.
I use the following command :
```
python3 tst_luma.py --display st7735 --interface=spi --spi-bus-speed=16000000 --gpio-reset=24 --gpio-data-command=23 --gpio-backlight=18 --width=160 --height=128 --backlight-active=high --rotate=3
```

I'd like to define the rotation of the screen in my program.
I tried these following lines instead of get_device() but it doesn't work. Nothing is happening.
```
from luma.core.interface.serial import spi
from luma.lcd.device import st7735
serial = spi(port=0, device=0, gpio_DC=23, gpio_RST=24, cs_high=True, bus_speed_hz=16000000)
device = st7735(serial_interface=serial, width=160, height=128, rotate=3)
```
What am I doing wrong?

Thanks in advance

thijstriemstra, 2020-01-19:
@raf59 you probably need to specify amount of degrees rotation instead.

raf59, 2020-01-19:
Thanks for your answer @thijstriemstra !
I'm not sure to understand.
According to the doc, "rotate" option can accept only 0, 1, 2 or 3.
How would you specify the amount of degrees rotation ?
For the moment, even without defining a rotation, the screen doesn't turn on if I don't get the "device" object using get_device().

rm-hull, 2020-01-19:
Try removing `cs_high=True,` when initializing SPI

raf59, 2020-01-19:
Oh thank you very much @rm-hull .
I don't know why I added this option. Probably a bad copy-paste.
Now the screen is working but the backlight keeps off.
I think it's due to the backlight-active option. I don't know how to pass this option to the device when initializing it.
The only way I found to turn the backlight on is to manually set the GPIO 18 high.
```
import RPi.GPIO as GPIO
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 1)
```
Is there a cleaner way to do it ?

rm-hull, 2020-01-19:
https://luma-lcd.readthedocs.io/en/latest/python-usage.html#backlight-control

rm-hull, 2020-01-19:
oh, you can pass `active_low=True|False` when initializing the device to indicate whether the backlight is active low or high

raf59, 2020-01-21:
Perfect !!!
Now, everything is working fine ! 
Thank you for your answer and also for this awesome library. Good job!

closed 2020-01-21 by raf59