rm-hull/luma.lcd #103: ST7567  1.8 LCD

XipoZhang, 2020-02-20:
Dear All:

the system cannot find the  following lib. Could someone give me a hint?  
  import pytest
  | from baseline_data import get_reference_data, primitives
  | from helpers import Mock, serial, call, setup_function, assert_invalid_dimensions  # noqa: F401

The ST7567  1.8 LCD is not lighted up even I wired it as exactly what the document said. 
How to turn on the backlight? 
It mentioned that GPIO18 can be used for backligh> how to wire and code that?

Thanks a lot.
xipo

thijstriemstra, 2020-02-20:
can you share the code you're using?

XipoZhang, 2020-02-20:
Thanks for your feedback. All the codes are from the the examples here. There is nothing new.
I bought 3 OPEN-SMART 1.8INCH LCD ST7567. None of them has any response. 
So my first question now is how to light up to check whether it works?

The GPIO setup
Board - GPIO Pin# (GPIO name)

GND	- 	09
3V3	-	01
SDI	-	19 (GPIO10 SPI0-MOSI)
SCK	-	23(GPIO11 SP10_CLK)
DC	-	16(GPIO23)
CS	-	24(GPIO08, SPI0-CE0_N)
RST	-	18(GPIO24)	

Could you please help me to check whether the wiring is right?