rm-hull/luma.lcd #107: ili9341 issues

starmood, 2020-05-16:
#### Type of Raspberry Pi
Pi3B with Buster
python 3.7
luma.lcd 2.3.0
#### Linux Kernel version
4.19.97-v7+

#### Actual behaviour
Found some strange behaviour.
For example.

1) The position of the graphic is not stable.
For testing, if I create a rectangle with the LCD width and height (320*240), sometimes it just likes being added some offset, I may not be ablt to see the left edge of the rectangle. But if I run the script again, it may just shows normal with all 4 sides (I did not set any h_offset or v_offset).

2) if I set h_offset (like set to 1), it's not be moved horizontally. Instead the graphic tilts 45 degrees, is this expected behaviour?

thijstriemstra, 2020-05-16:
can you attach a picture or make video of the behaviour?

starmood, 2020-05-16:
[image]
[image]
[image]

I use Pillow for testing:

First picture is nomal one.
Second one is sometimes happened, left edge does not show
Thired one is h_offset = 1

rm-hull, 2020-10-25:
@starmood can you share the program code that causes this please?

starmood, 2020-10-31:
Hi rm-bull, 

@rm-hull 

Below is the test code,
You should see not all the edge of rectangle show. 
Then you can change h_offset to 1 to check the weird transforming.
==============================================================
from luma.lcd.device import ili9341
import RPi.GPIO as GPIO
from luma.core.interface.serial import spi    
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont

gpiodc = 25
gpiorst = 27
gpiobl = 20
rotate = 0
v_offset = 0
h_offset = 0

serial = spi(bus_speed_hz=32000000, transfer_size=4096, port=0, device=0, gpio_DC=gpiodc, gpio_RST=gpiorst)  
device = ili9341(serial,rotate=rotate,h_offset=h_offset,v_offset=v_offset,gpio_LIGHT=gpiobl)

device.clear()

oled_width = device.width
oled_height = device.height

image       = Image.new('RGB', (oled_width, oled_height))
draw        = ImageDraw.Draw(image)

while true:
	draw.rectangle((0,0,oled_width,oled_height), outline = (255,0,0))
	device.display(image)

rm-hull, 2020-10-31:
The device width is 320 pixels and height is 240 pixels.

If you want to draw a border box, it should be (0, 0, 319, 239) ie use`oled_width - 1` and `oled_height - 1` When you are drawing the rectangle

starmood, 2020-10-31:
But I did not have same issue with other LCD screen, only 9341. And if I use Adafruit module, also not such issue.
As such, some complicated drawing result became strange with same code.
And when set h_offset, it also does not show as expected.

avjui, 2021-09-24:
Same issues here. After debug a while i found a fix. It seems it need some time after configuration. Same line also insert in the Adafruit library

Please can you add a sleep to this lines
https://github.com/rm-hull/luma.lcd/blob/9a58dab76c7fcc97089a2b9df88d6e09ac639b72/luma/lcd/device.py#L670-L674
```                     0x48, 0x08, 0x0f, 0x0c, 0x31, 0x36, 0x0f)
        self.command(0x11)                                # Sleep out
        sleep(0.12)
        self.clear()
        self.show()
```
Thanks

thijstriemstra, 2021-09-25:
thanks @avjui, can you make a pull request with that change?

avjui, 2021-09-27:
> thanks @avjui, can you make a pull request with that change?

Done!

thijstriemstra, 2021-11-14:
Fixed with #146 but not released yet. Could you help there @rm-hull?

closed 2021-11-14 by thijstriemstra