rm-hull/luma.lcd #111: How to print sensor data to nokia 5110 lcd?

pkrana0005, 2020-07-28:
I want to print sensor data to the nokia 5110 lcd using RPi zero w. I have seen the examples but unable to fiqure out to print data assigned to variables. I don't know the syntax. I saw syntax to print a string.

I am adding here my test code

import time
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.lcd.device import pcd8544
from RPi import GPIO

serial = spi(port=0, device=0, gpio_DC=23, gpio_RST=24)

device = pcd8544(serial,rotate=0)
device.contrast=0

with canvas(device) as draw:
       draw.text((0,0),text="hi", fill=255)

for j in range(0,100,10):
    # I want to print "j" value to the nokia 5110 lcd,
    #i dont know the syntax to print it
    print("j= ",j)
    time.sleep(2)

time.sleep(3)

pkrana0005, 2020-07-28:
I found the answer

for j in range(0,100,10):
    # I want to print "j" value to the nokia 5110 lcd,
    #i dont know the syntax to print it
    with canvas(device) as draw:
            draw.text((5,5), 'j={}' .format(j), fill=255)   # by writing this line i can print j to lcd
    print("j= ",j)
    time.sleep(2)

closed 2020-07-28 by pkrana0005

rm-hull, 2020-07-28:
Glad you managed to solve it !