rm-hull/luma.lcd #112: Need assistance with Unicode to font table mappings for the hd44780 class

dhrone, 2020-08-28:
The new hd44780 driver contains embedded fonts.  The data for these fonts are contained within the hd44780 class in luma.lcd.device.const.py in a dict called FONTDATA.  One key in FONTDATA is 'mappings' which is a list of two dicts (one for each of the two fonts that are typical for hd44780 style devices).  Each dict contains a collection of key value pairs which provide unicode values for as many of the table values as I could decode.  I'm looking for help to fill in the missing values and validate that I've correctly assigned the ones that are there.  I'd also be interested in any look-alike values that should be added (e.g. unicode codepoints that look the same as a mapping that already exists.  A good example of that would be the series of cyrillic mappings I've included for their english equivalents.  

Here's a small example of what the tables looks like...
```        'mappings': [
            {   # A00 ENGLISH_JAPANESE CHARACTER FONT
                # Missing maps for
                # a1, e2, e5, e6, e7, e9, ea, f0, f1, f4, f5, f8, f9, fa, fb, fc
                0x0410:   0x41,  # А CYRILLIC CAPITAL LETTER A
                0x00a5:   0x5c,  # ¥ YEN SIGN
                0x042c:   0x62,  # Ь CYRILLIC CAPITAL LETTER SOFT SIGN
                0x2192:   0x7e,  # → RIGHTWARDS ARROW
                0x2190:   0x7f,  # ← LEFTWARDS ARROW
                0x300c:   0xa2,  # 「 LEFT CORNER BRACKET
                0x300d:   0xa3,  # 」 RIGHT CORNER BRACKET
                0x30fd:   0xa4,  # ヽ KATAKANA ITERATION MARK
                0x30fb:   0xa5,  # ・ KATAKANA MIDDLE DOT
                0x30f2:   0xa6,  # ヲ KATAKANA LETTER WO
                0x30a1:   0xa7,  # ァ KATAKANA LETTER SMALL A
                0x30a3:   0xa8,  # ィ KATAKANA LETTER SMALL I
                0x30a5:   0xa9,  # ゥ KATAKANA LETTER SMALL U
                0x30a7:   0xaa,  # ェ KATAKANA LETTER SMALL E
                0x30a9:   0xab,  # ォ KATAKANA LETTER SMALL O
                0x30e3:   0xac,  # ャ KATAKANA LETTER SMALL YA
                0x30e5:   0xad,  # ュ KATAKANA LETTER SMALL YU
                0x30e7:   0xae,  # ョ KATAKANA LETTER SMALL YO
                0x30c3:   0xaf,  # ッ KATAKANA LETTER SMALL TU
}
```

The font tables are in the [hd44780 documentation](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf) on pages 17 and 18.

thijstriemstra, 2021-11-22:
@dhrone could you maybe provide a test script that scans through all the chars? how can we help?

fvzeppelin, 2022-03-26:
Not sure if this belongs here:
If I try to replace German Umlauts as described in the docs:
```
fnt = device.get_font('A02')
device.font.combine(fnt, ['Ä','Ö','Ü'])
device.text = 'ÄÖÜ'
```
Then the displayed result is surely not as expected. E.g. the Ä has a double horizontal bar in the middle and an additional horizontal bar at the bottom.