rm-hull/luma.lcd #118: HD44780 with I2C backpack

thijstriemstra, 2020-09-24:
I want to test out a HD44780 16 x 2 with an I2C backpack (e.g https://www.ebay.com/itm/PCF8574-I2C-Backpacks-for-HD44780-Compatible-LCD-Module-Ardiuno-Raspberry/281810234335). Looking at the docs, the I2C backpack is mentioned but none of the examples show how to use this: https://luma-lcd.readthedocs.io/en/latest/HD44780.html

Can you help @dhrone?

thijstriemstra, 2020-09-24:
Oh I see, the wiring part is here: https://luma-lcd.readthedocs.io/en/latest/hardware.html#hd44780-w-pcf8574

But no example of using it using code. https://luma-lcd.readthedocs.io/en/latest/HD44780.html could use a very visible link to the wiring page.

dhrone, 2020-09-24:
Unless you are interested in soldering as part of the exercise, I'd recommend buying displays with the PCF8574 already connected.  [Amazon Link](https://www.amazon.com/NOYITO-Backlight-Interface-Arduino-MEGA2560/dp/B07SYPPTWN/)

I took a quick look at the documentation for this display and it looks like it is wired to the defaults I've used for the pcf8574 class.  If that is true, then the following example should work...

```
from luma.core.interface.serial import pcf8574
from luma.lcd.device import hd44780

interface = pcf8574(address=0x27)
device = hd44780(interface)
device.text = 'Hey, this works'
```

thijstriemstra, 2020-09-24:
> Unless you are interested in soldering as part of the exercise, 

Not really but if it keeps the price to a minimum, no problem.

I already soldered this one:

[image]

[image]

How would I control the backlight?

dhrone, 2020-09-24:
The standard wiring has the backlight connected to PCF8574 pin D3 which is the default in the PCF8574 class.  If your wiring connected it to a different pin, you'll need to provide that during initialization.

`interface = pcf8574(address=n, BACKLIGHT=m)`
where n is the I2C address (default 0x27) and m is the data line on the backpack that the backlight is connected to (default 3).

thijstriemstra, 2020-09-24:
Ah there's also a lot of info in #90. Will close this and add comments to that ticket if something's not working.

closed 2020-09-24 by thijstriemstra

thijstriemstra, 2020-09-24:
Ended up with:

```python
import time

from luma.core.interface.serial import pcf8574
from luma.lcd.device import hd44780

interface = pcf8574(address=0x27, backlight_enabled=True)
device = hd44780(interface, width=16, height=2)
device.text = "Hello world"

time.sleep(20000)
```