rm-hull/luma.lcd #122: test_ili9341 hangs

thijstriemstra, 2020-10-16:
Output of `tox -r -e py38`

```
================================================================ test session starts ================================================================
platform linux -- Python 3.8.5, pytest-4.5.0, py-1.9.0, pluggy-0.13.1 -- /home/thijs/projects/photobooth-deploy/src/luma.lcd/.tox/py38/bin/python
cachedir: .tox/py38/.pytest_cache
rootdir: /home/thijs/projects/photobooth-deploy/src/luma.lcd
plugins: cov-2.10.1
collected 58 items                                                                                                                                  

tests/test_backlight.py::test_unsupported_platform PASSED                                                                                     [  1%]
tests/test_backlight.py::test_init PASSED                                                                                                     [  3%]
tests/test_backlight.py::test_gpio_cleanup PASSED                                                                                             [  5%]
tests/test_backlight.py::test_pwm_cleanup PASSED                                                                                              [  6%]
tests/test_backlight.py::test_pcf8574_cleanup PASSED                                                                                          [  8%]
tests/test_backlight.py::test_active_low_enable_on PASSED                                                                                     [ 10%]
tests/test_backlight.py::test_active_low_enable_off PASSED                                                                                    [ 12%]
tests/test_backlight.py::test_active_high_enable_on PASSED                                                                                    [ 13%]
tests/test_backlight.py::test_active_high_enable_off PASSED                                                                                   [ 15%]
tests/test_backlight.py::test_pwm_turn_off PASSED                                                                                             [ 17%]
tests/test_backlight.py::test_pwm_turn_on PASSED                                                                                              [ 18%]
tests/test_backlight.py::test_pwm_turn_50_percent PASSED                                                                                      [ 20%]
tests/test_backlight.py::test_pwm_unsupported_platform PASSED                                                                                 [ 22%]
tests/test_hd44780.py::test_init_4bitmode PASSED                                                                                              [ 24%]
tests/test_hd44780.py::test_init_8bitmode PASSED                                                                                              [ 25%]
tests/test_hd44780.py::test_display PASSED                                                                                                    [ 27%]
tests/test_hd44780.py::test_custom_full PASSED                                                                                                [ 29%]
tests/test_hd44780.py::test_get_font PASSED                                                                                                   [ 31%]
tests/test_hd44780.py::test_no_contrast PASSED                                                                                                [ 32%]
tests/test_hd44780.py::test_i2c_backlight PASSED                                                                                              [ 34%]
tests/test_hd44780.py::test_i2c_does_not_support_backlight PASSED                                                                             [ 36%]
tests/test_hd44780.py::test_unsupported_display_mode PASSED                                                                                   [ 37%]
tests/test_ht1621.py::test_init_6x8 PASSED                                                                                                    [ 39%]
tests/test_ht1621.py::test_cleanup PASSED                                                                                                     [ 41%]
tests/test_ht1621.py::test_hide PASSED                                                                                                        [ 43%]
tests/test_ht1621.py::test_show PASSED                                                                                                        [ 44%]
tests/test_ht1621.py::test_display PASSED                                                                                                     [ 46%]
tests/test_ili9341.py::test_init_320x240 PASSED                                                                                               [ 48%]
tests/test_ili9341.py::test_init_240x240 PASSED                                                                                               [ 50%]
tests/test_ili9341.py::test_init_320x180 PASSED                                                                                               [ 51%]
tests/test_ili9341.py::test_init_invalid_dimensions PASSED                                                                                    [ 53%]
tests/test_ili9341.py::test_offsets PASSED                                                                                                    [ 55%]
tests/test_ili9341.py::test_contrast PASSED                                                                                                   [ 56%]
tests/test_ili9341.py::test_hide PASSED                                                                                                       [ 58%]
tests/test_ili9341.py::test_show PASSED                                                                                                       [ 60%]
tests/test_ili9341.py::test_display 
```

thijstriemstra, 2020-10-16:
We should use `pytest-timeout` to ensure there's a reasonable timeout, or it might hang for hours, with e.g. `pytest --timeout=3`.

rm-hull, 2020-10-17:
Does it consistently timeout for you? 
What platform?

thijstriemstra, 2020-10-17:
> Does it consistently timeout for you?
Yes.

> What platform?

Ubuntu 20. But also on Travis @rm-hull: https://travis-ci.org/github/rm-hull/luma.lcd/jobs/736192575#L798

PR #123 (2020-10-18): tests: add timeout using pytest-timeout
fixes #122
closed 2020-10-18 by rm-hull