rm-hull/luma.lcd #143: Multiple TFT Screens (ST7735)

TheBlackSheep, 2021-07-14:
Hopefully posting in the right place...

Running on Raspberry Pi 2B+ running latest RaspOS lite

I have two ST7735's (they are both AZDelivery red boards and I think these are ST7735S) connected via SPI - I've tried various combinations of wiring - currently the first device is as per the recommended wiring , the second is on CE1 (GPIO7) for the CS line and I have separate DC (GPIO 20) and RST (GPIO16) but I have tried using the same DC and RST as the first screen.

I've also pinned the backlight to 3.3V - I couldn't get pin 18 to control them so have commented out that bit in my source.

this is the Linux version
5.10.48-v7+ #1435 SMP Fri Jul 9 18:18:40 BST 2021 armv7l GNU/Linux

I am trying to get graphics appearing on each independent display

this is the code I'm currently on;

```
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.lcd.device import st7735
import RPi.GPIO as GPIO

GPIO.setwarnings(False)

serial = spi(port=0, device=0, gpio_DC=23, gpio_RST=24, bgr=True)
serial1 = spi(port=0, device=1, gpio_DC=20, gpio_RST=16, bgr=True)
device = st7735(serial)
device1 = st7735(serial1)

#device.backlight(True) #turns the backlight on for both screens
#device1.backlight(True)

with canvas(device) as draw:
	draw.rectangle(device.bounding_box, outline="white", fill="black")
	draw.text((30, 40), "Device 0", fill="red")

with canvas(device1) as draw:
	draw.rectangle(device1.bounding_box, outline="white", fill="black")
	draw.text((30, 40), "Device 1", fill="blue")
```
I've suppressed the warnings as when the screens use common RST and/or DC lines I get a "channel in use" warning.  

The displays do show the text but quickly reset back to just the backlight - i.e. plain white.  I have had the first screen show ok and the second briefly appears but never stays.

Anyone any idea if this is possible?

Chris

thijstriemstra, 2021-07-14:
> The displays do show the text but quickly reset back to just the backlight - i.e. plain white

Add a `time.sleep(10)` at the end of the script?

TheBlackSheep, 2021-07-14:
Great that works! - hadn't expected the library to "cleanup" the screen(s) on exit.

Thanks

Chris

closed 2021-07-14 by TheBlackSheep