rm-hull/luma.lcd #147: Support ILI9488 3.5" (320x480) displays

TCB13, 2021-09-28:
Hello,

Are there any plans to support for 3.5" (320x480) displays with the ILI9488 chip? Or anyone already did it?

Examples of devices:

- http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520
- https://pt.aliexpress.com/item/32985467436.html?algo_exp_id=d2a973dd-545d-4a92-8b61-e427dcc58f4c-1&pdp_ext_f=%7B%22sku_id%22%3A%2266802769934%22%7D

Thank you.

rm-hull, 2021-09-29:
No plans to write a driver for it, but if there is a Linux framebuffer driver for this device, you could use https://luma-core.readthedocs.io/en/latest/device.html#luma.core.device.linux_framebuffer instead

TCB13, 2021-09-29:
Interesting, so I can get a `PIL.ImageDraw` out of the core device and use it to draw stuff in the same way I do with supported screens?

Side question, this library seems to support ILI9341 and there's also fb_ili9341 for it. Why is supported natively by the library and the other is via framebuffer? Or it all comes down to nobody with the skills to implement it on the library needs it?

Thank you.

rm-hull, 2021-09-29:
> Interesting, so I can get a PIL.ImageDraw out of the core device and use it to draw stuff in the same way I do with supported screens?

Yes, exactly

> Side question, this library seems to support ILI9341 and there's also fb_ili9341 for it. Why is supported "natively" by the library and the other is via framebuffer? 

The original ILI9341 driver was quite slow - pushing out a lot of pixels in python on a RPi was "challenging". We made some improvements to the diff to previous framebuffer, so it only drew changes since last time - this immensely improves FPS.

The framebuffer luma driver is universal (and was implemented after the ili9341 IIRC), and obviously there exists a number of different fb kernel drivers, so adding the framebuffer luma impl opened up to more devices. 

Also framebuffer driver is likely to be better performance for larger screen resolutions

> Or it all comes down to nobody with the skills and time to implement it on the library needs it?

I guess it mostly boils down to that yes, plus needing to acquire a device to test against

TCB13, 2021-09-29:
Thank your for the patience and detailed answer. It makes perfect sense now. I'm going to try the framebuffer driver route and see how it goes.

closed 2021-09-29 by TCB13