rm-hull/luma.lcd #153: The function device.backlight(False) turns on the lights, instead of device.backlight(True)

Kyuchumimo, 2021-11-26:
#### Type of Raspberry Pi
Raspberry Pi Zero W V1.1

#### Linux Kernel version
Linux raspberrypi 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux

#### Expected behaviour
Hardware: Nokia 5510
```
Pin  -  RPi GPIO (GPIO BOARD)
RST  -          18
CE   -          24
DC   -          16
DIN  -          19
CLK  -          23
VCC  -          01
BL   -          12
GND  -          06
```
```py
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.lcd.device import pcd8544
import RPi.GPIO as GPIO

serial = spi(port=0, device=0, gpio_DC=23, gpio_RST=24, gpio_LIGHT=18)
device = pcd8544(serial)

device.backlight(True)

try:
    while True:
        print(GPIO.input(18))
        
        with canvas(device) as draw:
            draw.text((0,0), "Hello World",fill=255)
except KeyboardInterrupt:
    GPIO.cleanup()
```
I was waiting for GPIO.input(18) to return 1 and the lights would turn on

#### Actual behaviour
GPIO.input(18) returns 0 and the lights do not turn on, but when changing the device.backlight(True) function to device.backlight(False) the lights turn on and GPIO.input(18) returns 1

thijstriemstra, 2021-11-26:
same happening with hd44780.

Kyuchumimo, 2021-11-27:
Using active_low=False as a parameter for the device object, makes the device.backlight(True) function turn the lights on and GPIO.input(18) returns 1. By default active_low parameter is always in boolean condition as True as long as it is not provided by the programmer.

```
device = pcd8544(serial, active_low=False)

device.backlight(True)
```

active low in electronics means a pin plugged to ground , and an active high a pin plugged to 5V or 3.3V.

From what I see this is normal. I suppose that for all LCD screens the backlight pin will not always be active low, as is the case with pcd8544 and hd44780 (without i2c backpack)

Please confirm if this is the case.

thijstriemstra, 2021-11-27:
> Please confirm if this is the case.

Not sure what you want me to confirm, but my hd44780 is connected to gpio pin 22, causing the backlight to be enabled all the time unless I call `device.backlight(True)`.

Kyuchumimo, 2021-11-27:
@thijstriemstra 
This is not how the backlight function works, backlight(True) should turn on the LED, not turn it off

see: https://luma-lcd.readthedocs.io/en/latest/python-usage.html#backlight-control
_The backlight can be programmatically switched on and off by calling device.backlight(True) or device.backlight(False) respectively._

thijstriemstra, 2021-11-27:
backlight(True) disables the backlight, which is incorrect imo, sorry if I wasn't clear. I would expect it the other way around like the docs state.

Kyuchumimo, 2021-11-27:
@thijstriemstra Wait, this is not a bug, it is expected. Haven't you read what I said?. The only thing I think is that active_low should be False by default by the luma.lcd library, since the backlight pin, which is almost always a positive pin leading to an LED must be active high for it to turn on

Kyuchumimo, 2024-05-27:
Tested on ST7735 with same behavior: device.backlight(False) makes turn on backlight LED

closed 2025-10-18 by rm-hull