rm-hull/luma.lcd #163: RuntimeError: This module can only be run on a Raspberry Pi!

frank-w, 2023-01-06:
Hi,

documentation says:

> luma.lcd provides a Python3 interface to small LCD displays connected to Raspberry Pi and other Linux-based single-board computers (SBC).

so this does not require raspberry-pi ;)

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

i try to get the lcd-lib working with spi/spidev on a bananapi and got this error (3.5inch rpi-display with 3v3 + 5v + gnd + spi-pins connected):

```
  File "/home/frank/.venv_display/lib/python3.9/site-packages/luma/core/lib.py", line 23, in __rpi_gpio__
    import RPi.GPIO as GPIO
  File "/home/frank/.venv_display/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

During handling of the above exception, another exception occurred:
```

#### Linux Kernel version
6.2-rc1

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

luma-lcd could access spidev already probed or i need to adjust pins/gpio-base? if yes where can i do this?

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

currently i get traceback because my device is no raspberry pi...how can i change this?

traceback above

regards Frank

rm-hull, 2023-01-06:
You would have to find a replacement library for RPi.GPIO (the low level GPIO library that luma uses). There is an OPi.GPIO for orange pi (which I am the main author of). I haven't kept track recently but many folks have used that for various different SBCs, but I don't know about banana pi. 

There have been small fixes but I haven't maintained it much lately: https://github.com/rm-hull/OPi.GPIO

frank-w, 2023-01-07:
have opened a thread in banana-pi forum

https://forum.banana-pi.org/t/bpi-r3-gpio-support-in-python/14662

there is a link in first posting pointing to bananapi-pi related gpio-lib which does not yet support bpi-r3

maybe your lib can support spidev? seen this in adafruit-library. this way no direct gpio-access is needed (except reset which can be done via sysfs or gpiolib). i'll find it strange that settings done in kernel already need to be done in userspace lib again.

i2c display is working (luma-oled module) without specific gpiolib through the i2c device in /dev maybe it is possible with spidev in same way :)

landall, 2024-09-12:
> You would have to find a replacement library for RPi.GPIO (the low level GPIO library that luma uses). There is an OPi.GPIO for orange pi (which I am the main author of). I haven't kept track [...]

so is there any api to load OPi.GPIO without RPi.GPIO?