rm-hull/luma.lcd #165: Comment error: HT1621 may actually 1/3 bias

xobs, 2023-03-10:
The following code appears to indicate that the HT1621 driver is using 1/2 Bias:

https://github.com/rm-hull/luma.lcd/blob/990716efc662210a265d8dbc14a8958da5723b8e/luma/lcd/device.py#L922

However, my understanding of the command format of luma.lcd is that this is actually setting 1/3 Bias:

[image]

The comment should be updated, or the code should be changed to 0x50 in order to use 1/2 Bias

thijstriemstra, 2023-06-11:
> The comment should be updated, or the code should be changed to 0x50 in order to use 1/2 Bias

@xobs did you try if it still works properly (or even better) when changing the code to use 0x50?

xobs, 2023-06-12:
For me, I had to use 1/3 bias in order to get the display to show anything. But the bias level depends on the display, so it varies from panel to panel.