rm-hull/luma.lcd #166: ILI9341 issue?

bbence84, 2023-03-12:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

Raspberry PI 4 4GB

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

Basically the ILI9341 does not work at all. I am trying to figure out if it's a HW issue (thought it does the same with 2 different units, althought they are from the same order). I just get a black screen, nothing else. I have used the default wiring as described here:
https://luma-lcd.readthedocs.io/en/latest/hardware.html#ili9341

I have this board:
https://www.amazon.co.uk/240x320-Screen-Serial-Module-ILI9341/dp/B07MXH92RL/

I have tried the following:

1. Init the board with explicit gpios specified:
spi_serial = spi(port=0, device=0, gpio_light=18, gpio_rst=24,  gpio_dc=23, gpio_cs=8, bus_speed_hz=52000000)  
self.device  = ili9341(spi_serial, width=320, height=240)   

2. Init the board without explicit gpios
3. Tried running one of the examples:  python matrix.py --config ../conf/ili9341.conf

In all cases, I get a blank black screen, no backlight, nothing.

However I can turn on the baclight with a simple python script:

import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
GPIO.setup(12, GPIO.OUT)
GPIO.output(12, 0)

But strangely, when I run the example with the ili9341 config or my own script with the default wiring from the wiki, I don't get anything, not even backlight turning on....

Any ideas? :)

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

Black screen, nothing displayed. I tripled checked all my wiring. And again, at least I would assume the blaclight to be turning on, but not even that is happening. Running a short script using RPi.GPIO works fine however. (It turns on the backlight but the screen still does not show anything)

Feuerwerko, 2025-01-03:
having the same problem

orkVili, 2025-01-15:
> having the same problem
I am trying to setup the display with Rpi5 have you figured it out yet?

bbence84, 2025-01-18:
@orkVili , @Feuerwerko : it was a while ago, but I ultimately ended up solving this (somehow), please refer to my Github repo: https://github.com/bbence84/pi_gptbot, and see the ILI9341 conf file that I am using now: https://github.com/bbence84/pi_gptbot/blob/main/app/ili9341n.conf, wiring: https://luma-lcd.readthedocs.io/en/latest/hardware.html#ili9341
Hopefully this will work.

orkVili, 2025-01-18:
Thank you!
I will check it out.