rm-hull/luma.lcd #178: Is the ILI9341V display supported?

nebhead, 2024-06-12:
#### Type of Raspberry Pi
Raspberry Pi Zero W (2W)

#### Linux Kernel version
Raspberry Pi OS Lite (Bullseye and Bookworm)

#### Expected behaviour
It appears that the ILI9341**V** might have some changes from the original ILI9341.  I'm using Luma LCD to support the ILI9341 displays in my project (https://github.com/nebhead/pifire) which has been working wonderfully for a couple of years.  However we have started to see this new variant of the ILI9341 pop-up in the market and they seem to have inverted colors.  

#### Actual behaviour
Colors should not be inverted on the display. 

[image]

thijstriemstra, 2024-06-12:
@nebhead [this esp-bsp ticket](https://github.com/espressif/esp-bsp/issues/238) mentions that "the difference is in initialization sequence" and shows the correct sequence. It's C code but maybe it helps to figure out what the difference is and how to implement it in python.

thijstriemstra, 2024-06-12:
And there's this driver https://github.com/juj/fbcp-ili9341/blob/master/ili9341.h that mentions 

> This has a subtle effect on colors/saturation.
> // Valid values are 0x20 and 0x30. Spec says 0x20 is default at boot, but doesn't seem so, more like 0x00 is default, giving 
> supersaturated colors. I like 0x30 best.
>// Value 0x30 doesn't seem to be available on ILI9340.

Perhaps it's these 'supersaturated colors' you're seeing?

nebhead, 2024-06-12:
I should also mention that I don't actually have one of these displays, but this was a report coming from a user of my project today. If we need to test any changes, maybe I can have this user help us with that. 

> And there's this driver https://github.com/juj/fbcp-ili9341/blob/master/ili9341.h that mentions
>
>
> Perhaps it's these 'supersaturated colors' you're seeing?

Maybe, but the background in the image should actually be black (and the text is white), so I think that it's inverse colors.  

Here is an example of what it should look like:

[image]